bark-simulator/bark #23: Running Bazel test for python examples

Jiraphapa, 2019-06-15:
Some test cases from running 'bazel test //...' are failed with TIMEOUT on macOS, it seems to be a problem with Python (virtual environment was activated). I've also tried to use --test_timeout on each failed test case but it seems to take forever. Below are the details
[Python Output.txt](https://github.com/bark-simulator/bark/files/3293555/Python.Output.txt)
 and the attachments: saved output from terminal and testlogs.

OS: macOS Mojave 10.14.5
Bazel version: Build label: 0.15.0
Python 3.7.3 (v3.7.3:ef4ec6ed12, Mar 25 2019, 16:39:00) 
[GCC 4.2.1 (Apple Inc. build 5666) (dot 3)] on darwin
Virtualenv version: 16.5.0

[Bark Terminal Saved Output.txt](https://github.com/bark-simulator/bark/files/3293535/Bark.Terminal.Saved.Output.txt)

[bazel-testlogs.zip](https://github.com/bark-simulator/bark/files/3293549/bazel-testlogs.zip)

[Python Output.txt](https://github.com/bark-simulator/bark/files/3293556/Python.Output.txt)

Jiraphapa, 2019-06-17:
So it's a problem with 'pybind11' type binding library on Python 3.7 on macOS. By changing configuration of "pybind11" from 2.2.3 to version 2.3.0 in WORKSPACE file did solve the issue.

klemense1, 2019-07-08:
updated WORKSPACE file for pybind version 2.3.0

klemense1, 2019-07-08:
moved to master

closed 2019-07-08 by klemense1