bark-simulator/bark #33: Missing PyGame GUI when running python examples

Jiraphapa, 2019-06-28:
running examples with commands like `bazel run //examples:od8_const_vel_one_agent` and `bazel run //examples:od8_const_vel_two_agent` encouters a missing GUI problem (only blank PyGame windows are shown), all tests are passed. Below are the details of the system:

OS: macOS Mojave 10.14.5
Bazel version: Build label: 0.15.0
Python 3.7.3 (v3.7.3:ef4ec6ed12, Mar 25 2019, 16:39:00)
[GCC 4.2.1 (Apple Inc. build 5666) (dot 3)] on darwin
Virtualenv version: 16.5.0

Jiraphapa, 2019-06-28:
adding  `pg.event.get()` after `pg.display.update()` in the function `show(self, block=True)` in `pygame_viewer.py` to register all events from the user into an event queue did solve the problem.

klemense1, 2019-07-08:
added to bark master

closed 2019-07-08 by klemense1