bark-simulator/bark #42: syntax error in WORKSPACE file

Jiraphapa, 2019-07-09:
when calling `bazel run` command, the following ERROR messages are given

> ERROR: /path/to/WORKSPACE:22:1: syntax error at '<': expected expression
> ERROR: /path/to/WORKSPACE:29:1: non-keyword arg after keyword arg
> ERROR: error loading package 'external': Failed to parse WORKSPACE file
> ERROR: error loading package 'external': Failed to parse WORKSPACE file
> INFO: Elapsed time: 0.093s
> INFO: 0 processes.
> FAILED: Build did NOT complete successfully (0 packages loaded)
> FAILED: Build did NOT complete successfully (0 packages loaded)

Jiraphapa, 2019-07-09:
adding `#sha1 = "8edc147d67ca85a93ed1f53628004528dc36a04d"` for pybind11 did solve the problem:

> http_archive(
>     name = "pybind11",
>     #sha1 = "8edc147d67ca85a93ed1f53628004528dc36a04d",
>     strip_prefix = "pybind11-2.3.0",
>     urls = ["https://github.com/pybind/pybind11/archive/v2.3.0.zip"],
>     build_file = "@//tools/pybind11:pybind.BUILD",
> )
>

klemense1, 2019-07-11:
This seems odd, as #sha1 is just a commend. I would assume that it should not change anything to add such a line. Also, the problem does not occur on the ci and under linux. You're os is mac, right?

Jiraphapa, 2019-07-11:
yes, it's macOS 10.14.5

klemense1, 2019-07-11:
Alright, I will try it out again to reproduce it.

klemense1, 2019-07-13:
I could not reproduce the issue, thus I am going to close it for now.

closed 2019-07-13 by klemense1