bark-simulator/bark #47: FindNearestLanes doesn't return the correct lanes

ctonic, 2019-07-18:
I commented out the part from xodr_parser where all the lanes other than of "driving" type are ignored, so the rtree in MapInterface should have all lanes.

Then I queried the 3 nearest lanes from a given point using (the python binding) of FindNearestLanes.

For this point I would expect the 3 nearest lanes to be of type sidewalk (5), border (6) and driving (1):
[image]

But the function returns two lanes of type 1 and one of type 6:
[image]

When trying it for a different point:
[image]
Here the function works. It returns (as expected) lanes of type none (0), border (6) and sidewalk (5).

I also tried it with random other points along the road, for some points it gives the expected results, for others not.

I circumvented this problem by creating my own function to find the nearest lanes which uses:
`boost::geometry::index::satisfies([&](rtree_lane_value const& v)
{return distance(v.second->get_line(), point) <= radius; }),` 

instead of `boost::geometry::index::nearest(point, num_lanes);`.
This returns the expected results (but requires a radius parameter).

I also created a function to find all the lanes that are intersected by a given line and when doing the rtree query using `boost::geometry::index::intersects(line_segment),` it didn't return anything when it clearly should. Using a own lambda (here using modules::geometry Collide) solved this problem too.

ctonic, 2019-07-18:
I did a quick runtime comparison and using bgi::satisfies with distance (the way I implemented it) is more than 10 times slower than the version using bgi::nearest so it's not really an alternative.

AKreutz, 2019-09-03:
I have another issue with FindNearestLanes. I want to use the function to find the outer boundary of the corridor that an agent currently is located in. Intuitively this could be done by calling `FindNearestLanes(agent_position, 2, lanes)`. This returns two lanes, and the one with the higher absolut lane position should be the outer boundary.
This approach however fails, if the vehicle is located between the lanes with lane positions 0 and 1, because FindNearestLane never returns the lane with position 0 (i.e. the plan view). In this case, the function would return lanes 1 and 2, which would place the agent in the wrong corridor.

My workaround is to compare the distance between the vehicle position and the outer lane with the distance between the inner and outer lanes to detect this situation: 
```
if (geometry::distance(outer_lane->get_line(), position) > geometry::distance(inner_lane->get_line(), outer_lane->get_line())) {
  // Position is between lane 0 and 1, but the two nearest lanes are 1 and 2, because 0 is never returned -> current_lane is the inner_lane
  return inner_lane;
} else {
  return outer_lane;
}
```
This workaround could fail if the all lanes do not have the same offset to each other.
@klemense1

patrickhart, 2019-09-13:
Does this issue still exist? @AKreutz @ctonic @klemense1

AKreutz, 2019-09-16:
My comment is not really an issue with Bark but a functionality that would be nice to have.
"Finding the outer boundary Line of the lane a vehicle is currently driving in" is a use case that I have stumbled upon twice, without having found a simple way of implementing it. I just wanted to mention it so that you can decide if this procedure is something that would be useful to have in the future.

klemense1, 2019-11-29:
should be solved by commit https://github.com/bark-simulator/bark/commit/dcbf0fd3a5d5a8f080184c57d71abab4fd8b5c52#diff-22c74272bd958ea74cb21f3612cea70f
( changing rtree from outer lane line to center line )

closed 2019-11-29 by klemense1