bark-simulator/bark #73: Bug: Lanes are sometimes not created with the correct length

AKreutz, 2019-08-29:
When a Lane is created using the function `modules::world::create_lane_with_offset_from_line` in [commons.hpp](https://github.com/bark-simulator/bark/blob/8a963e9124de38ca10a5cac65d0c3e133f02a288/modules/world/opendrive/commons.hpp#L165), the created line sometimes is too short by one point.

The bug happens when the requested length of the new Lane, as determined by the parameters `lane_width_current_lane.s_start` and `lane_width_current_lane.s_end`, is not equal to 1.
This is almost always the case when a map is created using [xodr_parser.py](https://github.com/bark-simulator/bark/blob/8a963e9124de38ca10a5cac65d0c3e133f02a288/modules/runtime/commons/xodr_parser.py#L317).

AKreutz, 2019-08-29:
Pull request #74 contains a test that fails because of the bug, as well as a potential fix for it.

klemense1, 2019-09-20:
- cleaned up create_lane_with_offset_from_line in https://github.com/bark-simulator/bark/pull/122
- sampling to the exact length creates problems with curves, as the previous line is also a curve --> a inner lane has to be shorter and a outer lane has to be longer.
- sampling lines not up to the full length does imho no harm, as linestrings are concatinated for planning anyway. 
- it's probably most obvious in plotting, but we can live with that.
- i would close the ticket

closed 2019-09-20 by klemense1