bark-simulator/bark #75: Bazel Version 0.29.0 not supported

klemense1, 2019-09-02:
`ERROR: /home/esterle/.cache/bazel/_bazel_esterle/686875265778f8801c25fece24de0587/external/planner_uct/src/BUILD:1:1: no such package '@bark_project//modules': BUILD file not found in directory 'modules' of external repository @bark_project and referenced by '@planner_uct//src:planner_uct_single_agent'
ERROR: Analysis of target '//modules/models/behavior/plan:uct_single_agent' failed; build aborted: no such package '@bark_project//modules': BUILD file not found in directory 'modules' of external repository @bark_project
INFO: Elapsed time: 0.296s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (0 packages loaded, 0 targets configured)
    Fetching @com_github_eigen_eigen; fetching
    Fetching @gtest; fetching
    Fetching @boost; fetching
`

klemense1, 2019-09-02:
- pushed current (broken) master to branch planner_submodule_bug
- fixed master for now by commenting out @planner_uct//src:planner_uct_single_agent in behavior/plan BUILD file

juloberno, 2019-09-10:
Now external planners are built only if - - define planner_uct=true is added to the bazel call. Problems with planner repositories thus do not influence the main bark build

closed 2019-09-10 by juloberno