bark-simulator/bark #81: Reduce Compilation Times of Ceres Build

klemense1, 2019-09-06:
Ceres Build takes most of the time of our bild time.
I've implemented a BUILD file for the ceres library where the option restrict_schur_specializations is set to true. However, it reads in all srcs and headers at once from bazel/ceres.bzl (in the ceres repository). Maybe dividing this into several libraries as in https://github.com/nelhage/rules_boost/blob/master/BUILD.boost would speed up the build process.

juloberno, 2019-09-10:
Only build if mpc required?

As implemented with planners, adding - - define mpc=true to build the execution model only if required.

klemense1, 2019-10-07:
I prefer a solution similar to boost. Chan will have a look into it.

klemense1, 2019-10-10:
- in modules/models/execution/mpc/mpc.hpp, we are including ceres/ceres.h
- in modules/models/execution/mpc/mpc.cpp, we are using Problem (from "ceres/problem.h"), DynamicAutoDiffCostFunction (from "dynamic_autodiff_cost_function.h") and Solver (from "ceres/solver.h")
--> maybe it helps if we specify these as separate libraries in the build file as it is done with boost (see link above)

as a benchmark, I would try bazel test //... and then bazel build //... --> use the time it takes for bazel build to benchmark the progress

tin1254, 2019-10-12:
I have separated Problem and Solver from the rest of the source files as two cc_library, however the total number of source files needed, is very difficult to trim down, as Problem and Solver solely need most of the source files.
The time it takes to compile the whole project is the same as before, takes around 5 mins, and compiling ceres takes only a small fraction of time by my observation...

klemense1, 2019-10-14:
can you push it to some branch so I can have a look into it?

tin1254, 2019-10-14:
Yes, it is in here:
https://github.com/bark-simulator/bark/blob/tin1254-bug-fixing/tools/ceres/ceres.BUILD

klemense1, 2019-10-14:
Can you elaborate why you choose to include all headers using glob? My initial idea was to only include the necessary headers?
maybe a good reference could be boost::geometry from the repo above. It includes libraries again such as algorithms, and so on. What do you think?

tin1254, 2019-10-14:
My thought to the headers is, the irrelevant headers will be ignored during the parsing phase, as they are not included in the target source code files, so I guess the extra time would be neglectable.
I see, my original thought was to just pick the source files we need while separating Problem and Solver as in boost::geometry. I will try to just separate them.

klemense1, 2019-10-14:
Isn't glob searching for all headers in the directory? Can't you specify the headers just like you did with the source files?

tin1254, 2019-10-14:
OK, I will try to do that too.

klemense1, 2019-10-17:
#144 does not really improve the build time. building and testing in circle ci takes pretty much the same time (~32).

klemense1, 2019-10-17:
without Ceres and the Execution MPC: Elapsed time: 351.773s, Critical Path: 46.10s
with Ceres (current master branch): Elapsed time: 435.619s, Critical Path: 48.11s

klemense1, 2019-10-29:
with #144 , we treat this issue as soved

closed 2019-10-29 by klemense1