bark-simulator/bark #97: Problem with Polygon2d

patrickhart, 2019-09-11:
The polygon has to be post translated and cannot be initialized with correct center. At least plotted wrong.
`
      goal_polygon = Polygon2d([0, 0, 0],
                               np.array(agent_json["goal"]["polygon_points"]))
      goal_polygon = goal_polygon.translate(Point2d(agent_json["goal"]["center_pose"][0],
                                                    agent_json["goal"]["center_pose"][1]))
`

xmyqsh, 2019-09-11:
@patrickhart Besides the problem you report, there is another problem about `Ploygon2d.translate`.
The return type of `modules::geometry::Ploygon:: translate` is `boost::geometry::model::Ploygon`, not `modules::geometry::Ploygon`.
Also, there is no pybind porting for the type of boost::geometry::model::Ploygon`.

xmyqsh, 2019-09-11:
@patrickhart Another problem, the return type of `goal_polygon.translate(` is `Shape<bg::model::polygon<T>, T>`, not `Polygon`.
And, `Shape<bg::model::polygon<T>, T>` has not been ported to python by pybind.

xmyqsh, 2019-09-11:
@patrickhart 
Also, I feel the design of `geometry` is not as good as other parts of the repo.

xmyqsh, 2019-09-11:
A better solution:
```
+ 106 template <typename T>
+ 107 inline boost::shared_prt<Polygon_t<T>> Polygon_t<T>::translate(const Point2d &point) const {
+ 108   return boost::shared_prt<Polygon_t<T>>(std::dynamic_cast<Polygon_t<T>*>(Shape<bg::model::polygon<T>, T>::translate(point).get()));
+ 109 }
+ 110
```
Then, port `boost::shared_prt<Polygon_t<T>>` with pybind. Replace by `boost::shared_prt<` on the relevant part.

xmyqsh, 2019-09-11:
@patrickhart 
For the problem you proposed, #96
How about this fix patch?
```
   48 template<typename T>
   49 inline Polygon_t<T>::Polygon_t(const Pose &center, const std::vector<T> points) :
   50            Shape<bg::model::polygon<T>, T>(center, points, 0),
+  51            auto tmp = dynamic_cast<Polygon_t<T>*>(translate(center));
+  52            swap(this, tmp);
+  53            delete tmp;
   54            rear_dist_(0.0f),
   55            front_dist_(0.0f),
   56            left_dist_(0.0f),
   57            right_dist_(0.0f) {
   58     boost::geometry::correct(Shape<bg::model::polygon<T>, T>::obj_);
   59     UpdateDistancesToCenter();
   60 }
   61
```

xmyqsh, 2019-09-11:
A better solution:
```
   49 inline Polygon_t<T>::Polygon_t(const Pose &center, const std::vector<T> points) :
   50            Shape<bg::model::polygon<T>, T>(center, points, 0),
   51            rear_dist_(0.0f),
   52            front_dist_(0.0f),
   53            left_dist_(0.0f),
   54            right_dist_(0.0f) {
+  55     bg::strategy::transform::translate_transformer<double, 2, 2> translate_backwards(center_[0], center_[1]);
+  56     boost::geometry::transform(obj_, obj_, translate_backwards);
   57     boost::geometry::correct(Shape<bg::model::polygon<T>, T>::obj_);
   58     UpdateDistancesToCenter();
   59 }
```

xmyqsh, 2019-09-11:
https://github.com/bark-simulator/bark/pull/96/files/56a5e64cb6c19a42aa8f59160c49a1dde6a9ab93..62dbaea90ec63f0f2f4c92c99936354c31cc33ee
This HACK is a bug because the `UpdateDistancesToCenter();` has not been called in the prop place.

patrickhart, 2019-09-11:
I will look into it tomorrow. For now the hack has to be sufficient.

xmyqsh, 2019-09-11:
@patrickhart 
I feel the original implement is correct. Why use the wrong `HACK`?

patrickhart, 2019-09-11:
> @patrickhart
> I feel the original implement is correct. Why use the wrong `HACK`?

It for sure does not produce the right outcome. I will have a look tomorrrow

juloberno, 2019-09-11:
The polygon should not be translated to the center initially, the center is always a fixed point of translation and rotation for a polygon. This is the concept of the shape.

@xmyqsh Better replace it with std::shared_ptr, the return of the translate function, or even better unique_ptr but this may be problematic with pybind

patrickhart, 2019-09-11:
> The polygon should not be translated to the center initially, the center is always a fixed point of rotation for a polygon.t that's why the center is a pose to give the initial 0degree orientation of the shape. You can have the center outside of the polygon also. Check the geometry python tests, where the current implementation gets clear.
> 
> @xmyqsh Better replace it with std::shared_ptr, the return of the translate function, or even better unique_ptr but this may be problematic with pybind

Alright, in that case I just remove the HACK and it can be considered as fixed.

closed 2019-09-11 by patrickhart