bark-simulator/bark #112: Make Python bindings congruent

patrickhart, 2019-09-17:
Avoid a mixed usage of def_property(..) and def(..).
Furthermore, the naming should also be made congruent.

patrickhart, 2020-01-09:
We should agree on how we want to define properties, functions and classes.

klemense1, 2020-01-23:
Rule:
- As long as the function has no arguments, it should be accessed by property.
- otherwise wrap the function

juloberno, 2020-01-24:
What about setters and getters (setters do have an argument)? 
I suggest as rule: Properties in python should be directly linked to class members in c++

patrickhart, 2020-01-24:
C++ functions have been renamed using CamelCase.
I propose we do the same for python.
Agree with the above-proposed ideas.

patrickhart, 2020-01-25:
Adapted Python bindings are on branch `python_bindings`.
Have a look and check if these are complete.

PR #234 (2020-01-27): Adapted Python bindings
Resolves #112
closed 2020-01-27 by patrickhart