bark-simulator/bark #141: intersection scenario: driving corridor und routing läuft falsch

juloberno, 2019-10-08:
[image]

scenario generation uniform vehicle distribution creates the attached scenario with drawn driving corridors for the agents. The routing and corridor generation for one agent fails.

juloberno, 2019-10-08:
example available on https://github.com/bark-simulator/bark/tree/intersection_scenario

klemense1, 2020-01-08:
is this still valid? we now have better tooling for analyzing the map.

patrickhart, 2020-01-08:
[image]
[image]

closed 2020-01-08 by patrickhart