bark-simulator/bark #167: Initial corridor generation

klemense1, 2019-10-29:
* Calculate RoutingCorridor
* Calculate RoadCorridor (Merge driving corridors as polygons)

klemense1, 2020-01-09:
links to #194

klemense1, 2020-01-09:
I think we should not book-keep any dynamically changing routing information DrivingCorridor. What is already implemented is that every agent tracks it's own CurrentLane.

I think what we also need is a new Lane class seperate from the OpenDrive Definition. Letting non-xodr-related stuff creep into that class is error-prone. And it also prevent's us from storing the lane-polygon directly in the lane class, which is why it's currently stored in the roadgraph. That means we currently always have to access the roadgraph for that.

Also, a new Lane-Class might allow us to store Right and Left Boundaries (and the repective road markings). The argument of driving direction is not suitable here, as we already have a definition of driving direction in the DrivingDirectionEvaluator.

patrickhart, 2020-01-23:
#223

closed 2020-01-23 by patrickhart