bark-simulator/bark #186: Incorrect test of driving corridor due to error in EvaluatorCollisionDrivingCorridor

AKreutz, 2019-12-05:
The test world_check_driving_corridor
https://github.com/bark-simulator/bark/blob/96c0201959940a43e4d2069047625910b44ef6dc/modules/world/tests/world_test.cc#L182-L203
does not correctly test if the agent collides with its DrivingCorridor. Because the agent is created without passing a MapInterfacePtr, the LocalMap and DrivingCorridor of the Agent are never generated.

The test only passes because of an error in EvaluatorCollisionDrivingCorridor:
https://github.com/bark-simulator/bark/blob/96c0201959940a43e4d2069047625910b44ef6dc/modules/world/evaluation/evaluator_collision_driving_corridor.cpp#L28-L31
Here, the `colliding` flag is set to `true`, if the agent polygon is **not** within the corridor polygon.

AKreutz, 2019-12-05:
The implementation of `Evaluate` in `EvaluatorCollisionDrivingCorridor` is correct, as the function is supposed to return `true` if the condition to be tested is not fulfilled.

klemense1, 2020-01-08:
Is this issue still relevant?

AKreutz, 2020-01-08:
There is no error in `EvaluateCollisionDrivingCorridor`. However, I think that the test case `world_check_driving_corridor` should be renamed because the purpose of the test is not clear from its name. The test does not check if a `DrivingCorridor` was correctly generated, it only confirms that it has not been generated yet.

patrickhart, 2020-01-23:
#223

closed 2020-01-23 by patrickhart