bark-simulator/bark #191: Issuees with edge connection of lane vertex in roadgraph

tin1254, 2019-12-10:
In GenerateFromJunctions, we may also need to consider `laneLink` under `connection` in `junction` of opendrive map.
And also decide whether `incomingRoad` & `connectingRoad`, or `predecessor` & `successor` implicate the possible driving direction, as we only add successor for those tags and the edges are directed now.
As a consequence, weird routing behavior occurs on Carla's maps.

tin1254, 2019-12-16:
[image]
[image]
There are consecutive lane sections connected to each other. Right now in roadgraph, we omitted this information and connect each of them directly to pre/successor.

To fix this, we might also want to consider the contactPoint tag.
`<predecessor elementType="road" elementId="0" contactPoint="end"/>`

klemense1, 2019-12-17:
in roadgraph.cpp, Line 449 `// TODO(@hart): there could be mult. lane_sections`
the issue with connecting multiple lane_sections  will probably be tackled at. what do you think? is that the problem we are facing with the junction?

tin1254, 2019-12-17:
> in roadgraph.cpp, Line 449 `// TODO(@hart): there could be mult. lane_sections`
> the issue with connecting multiple lane_sections will probably be tackled at. what do you think? is that the problem we are facing with the junction?

Those lines of code are the reason of another issue I encountered at another junction, but not the one I show in https://github.com/bark-simulator/carla-interface/issues/4#issuecomment-560503364.

tin1254, 2019-12-18:
> 
> 
> in roadgraph.cpp, Line 449 `// TODO(@hart): there could be mult. lane_sections`
> the issue with connecting multiple lane_sections will probably be tackled at. what do you think? is that the problem we are facing with the junction?

I think it would be great fix to bark, and it can surely alleviate the issues we faced on Carla maps

klemense1, 2019-12-19:
[Town01_small_v1.zip](https://github.com/bark-simulator/bark/files/3983996/Town01_small_v1.zip)

patrickhart, 2020-01-08:
#205

PR #205 (2020-01-08): Fixed Bug in the map generation and logging
The PlanView now contains only unique points.
Additionally, the functions of creating the PlanView and Lane lines have been refactored.
std::cout is now changed to LOG(info)

Resolves #191 #85 #19 #49
closed 2020-01-08 by patrickhart