bark-simulator/bark #193: get_all_neighbours on citiy highway map does not return all neighbours

cirrostratus1, 2019-12-17:
When calling 
https://github.com/bark-simulator/bark/blob/f234609804b76e7038af8a9b4b965659473bbaef/modules/world/map/roadgraph.cpp#L273
on the [city highway straight map](https://github.com/bark-simulator/bark/blob/f234609804b76e7038af8a9b4b965659473bbaef/modules/runtime/tests/data/city_highway_straight.xodr) for either lane, no neighbours are returned.

klemense1, 2019-12-18:
The roadgraph looks like this

[image]

That means the unit test does not fully make sense. The notion of lanes within the roadgraph (and it's neighbour properties) does not directly translate to our "driving domain"

klemense1, 2020-01-08:
@cirrostratus1 Is this issue still relevant?

cirrostratus1, 2020-01-08:
I think with the explanation you gave, this is the intended behavior, since the other lanes are in the opposite direction, so they probably should in general not be considered for a driving corridor (maybe if overtaking is allowed?). However, for clarification maybe the uct_planner example should be changed, since it uses the map as a highway with two lanes of the same driving direction.

patrickhart, 2020-01-09:
> I think with the explanation you gave, this is the intended behavior, since the other lanes are in the opposite direction, so they probably should in general not be considered for a driving corridor [...]

Is the driving direction clearly specified in the OpenDrive map? If not, I would close this issue.

klemense1, 2020-01-09:
> Is the driving direction clearly specified in the OpenDrive map? If not, I would close this issue.

We have agreed to a definition based on the sign of the lane position, which is used in https://github.com/bark-simulator/bark/blob/8294df8676f93db4137af742551d1d9f3adbac8e/modules/world/map/local_map.cpp#L37

patrickhart, 2020-01-23:
#223

closed 2020-01-23 by patrickhart