bark-simulator/bark #208: Simplify Linestrings when construncting Lanes

klemense1, 2020-01-09:
We currently construct the linestrings for arc and spiral from the Open-Drive Definition using a fixed spatial step-size (often 1m). However, that yields line-strings that may be quite complex.

Using Boost::Simplfy, we can simplify the line-string given an error-measure.

This would enhance speed for all sorts of post-usages:
- collision-checks of resulting polygons
- drawing / viewer

juloberno, 2020-01-24:
the boost simplify method also requires a parameter specifying the max_distance of vertices:

void simplify(Geometry const & geometry, Geometry & out, Distance const & max_distance)

We should clarify if we get any benefit, when use this method as postprocessing step after sampling a linestring with step-size 1m, instead of directly using a larger step size.

patrickhart, 2020-01-26:
At the moment even straight lines are sampled with 0.5m. Avoiding this could yield huge performance benefits.

PR #235 (2020-01-28): General improvements
- Deprecated lan.cpp functions
- Removed unnecessary python imports
Resolves #151 
Resolves #145
Resolves #215 
Resolves #208
closed 2020-01-28 by patrickhart