bark-simulator/bark #219: RoadGraph Extension

patrickhart, 2020-01-16:
The RoadGraph needs to have additional functionalities and data.

Requested functionalities:
- `RoadPtr GetNextRoad(road_id)`
-  Change Parser to include road connections
- `LanePtrGetLeftLane(lane_id)`
- `LanePtr GetRightLane(lane_id)`
- `Boundary GetLeftBoundary(lane_id)`
- `Boundary GetRightBoundary(lane_id)`
- `std::vector<XodrRoadId> GetRoadIdsForPath(road_start_id, road_end_id)`

patrickhart, 2020-01-16:
Mocked functions on branch road_corridor

klemense1, 2020-01-20:
Updated interfaces

```
  XodrLaneId GetNextLane(const XodrLaneId& lane_id)

  std::vector<XodrRoadId> FindRoadPath( const XodrRoadId& startid, const XodrRoadId& goalid)
```

We depricate   `XodrRoadId GetNextRoad(const XodrRoadId& road_id) {}`

patrickhart, 2020-01-23:
#223

closed 2020-01-23 by patrickhart