bark-simulator/bark #221: Map Interface Extension

klemense1, 2020-01-17:
Determine Left & Right Lanes based on the agent's pose
```

std::pair<LeftLanePtr, RightLanePtr> GetLeftAndLRightLane(pose) {
lane_id = FindXodrLane(pose)

float orientation_normalized = remainder(orientation, 2*M_PI);  // normalize to [0, 2pi]

float s = get_nearest_s(lane->get_line(), point);
float orientation_at_s = get_tangent_angle_at_s(lane->get_line(), s);
orientation_at_s = remainder(orientation_at_s, 2*M_PI);  // normalize to [0, 2pi]

bool same_orientation_as_road = (std::fabs(orientation_at_s - orientation_normalized) <= M_PI_2);

innerLane = getInnerLane(lane_id)

if (same_orientation_as_road) {
  leftLane = innerLane
  rightLane = currentLane
}
else {
  // flip them
}
}
```

klemense1, 2020-01-17:
also modify HasCorrectDrivingDirection() function to use features from map (insted lane_position definition as currently)

patrickhart, 2020-01-20:
Needs driving_direction as argument

patrickhart, 2020-01-23:
#223

PR #223 (2020-01-23): RoadCorridor Refactoring
In this refactoring we deprecated the LocalMap and DrivingCorridor.

The RoadCorridor is supposed to provide a more usable interface with more functionalities.

Resolves #221 #220 #219 #218 #210 #194 #193 #186 #167 #128 #118
closed 2020-01-23 by patrickhart