bark-simulator/bark #286: Python.h: No such file or directory

tom-doerr, 2020-02-27:
When I run `bazel build //...` I get the following error:
```
ERROR: /home/tom/git/bark/python/models/plan/BUILD:9:1: C++ compilation of rule '//python/models/plan:planners' failed (Exit 1) gcc-7 failed
: error executing command /usr/bin/gcc-7 -U_FORTIFY_SOURCE -fstack-protector -Wall -Wunused-but-set-parameter -Wno-free-nonheap-object -fno-omit-frame-pointer '-std=c++0x' -MD -MF ... (remaining 343 argument(s) skipped)

Use --sandbox_debug to see verbose messages from the sandbox
In file included from bazel-out/k8-fastbuild/bin/external/pybind11/_virtual_includes/pybind11/pybind11/pytypes.h:12:0,
                 from bazel-out/k8-fastbuild/bin/external/pybind11/_virtual_includes/pybind11/pybind11/cast.h:13,
                 from bazel-out/k8-fastbuild/bin/external/pybind11/_virtual_includes/pybind11/pybind11/attr.h:13,
                 from bazel-out/k8-fastbuild/bin/external/pybind11/_virtual_includes/pybind11/pybind11/pybind11.h:49,
                 from ./python/common.hpp:13,
                 from ./python/models/plan/plan.hpp:9,
                 from python/models/plan/plan.cpp:7:
bazel-out/k8-fastbuild/bin/external/pybind11/_virtual_includes/pybind11/pybind11/detail/common.h:112:10: fatal error: Python.h: No such file or directory
 #include <Python.h>
          ^~~~~~~~~~
compilation terminated.
INFO: Elapsed time: 8.189s, Critical Path: 6.99s
INFO: 2 processes: 2 linux-sandbox.
FAILED: Build did NOT complete successfully
```
I know that this happens when the `python-dev` package is missing, but on my system the `python3.7-dev` package is installed.

Proof:
```
➜  ~ apt list --installed | ag python3.7-dev

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libpython3.7-dev/bionic-updates,now 3.7.5-2~18.04 amd64 [installed]
python3.7-dev/bionic-updates,now 3.7.5-2~18.04 amd64 [installed]
➜  ~ 
```
I also installed the packages `python3.6-dev` and `python3.8-dev`, but that didn't fix it either.

Any ideas?

patrickhart, 2020-02-28:
Did you activate your virtual environment by using: source dev_into?

Also make sure that the Python.h is available in the python/venv folder. There seemed to be a bug in the latest virtualenv @juloberno?

tom-doerr, 2020-03-01:
`Python.h` is indeed not available in the `python/venv` folder. I copied a random `Python.h` file from my system into that location, but that didn't help. Any ideas how to fix the virtualenv?

patrickhart, 2020-03-02:
As I mentioned before: try rolling back to an older version of virtualenv. This might fix this issue.

juloberno, 2020-03-02:
Yes, Patrick is right. With version 16.0 it should work.

Install with pip install virtualenv==16.0.0 into system python.

klemense1, 2020-03-02:
for that, have a look in `tools/installers/install_python_venv.sh`

tom-doerr, 2020-03-02:
It works now, thank you all!

closed 2020-03-02 by tom-doerr