bark-simulator/bark #297: Fix WorldAtExecutionStep

juloberno, 2020-03-04:
WorldAtExecutionStep can calculate world states at intermediate  time steps, e.g. meaningful for smooth rendering of videos.

Currently, the execution model interpolation seems to be broken for intermediate execution steps.

Goal: Fix WorldAtExecutionStep, test with VideoRenderer, render_intermediate_steps=...

patrickhart, 2020-04-20:
@juloberno my suggestion would be to refactor the agent's Execute function and let this handle the execution model itself.
It would be way more neat, if the Execution model would return a state instead of a trajectory. However, what about the actions? Should these also be interpolated?

patrickhart, 2020-06-08:
Resolved in development branch with PR #364

closed 2020-06-08 by patrickhart