bark-simulator/bark #360: Add geoReference into xodr files and center the map to origin

tin1254, 2020-05-29:
Map issue encountered during integrating RSS:
1. [One of the dependences](https://github.com/carla-simulator/map) of RSS requires geoReference in the header to project points
2. In city highway straight, the road has the coordinates above 5000, which is invalid for the dependence of RSS. Centering the road to the origin solves the issue.

patrickhart, 2020-06-16:
Can this not be internally be done for the calculations of the RSS?
It is important for the map to remain at its position for applications such as dataset replays.

klemense1, 2020-06-18:
I don't think this will break dataset replay. @tin1254 can you please check if the german and chinese merging maps work?

patrickhart, 2020-06-19:
If it is only saved as semantic information then no. However, if the map is re-centered the coordinates will not match.

klemense1, 2020-06-19:
german and chinese merging maps have this already. the maps we made manually (city highway straight) are missing this information...

patrickhart, 2020-06-19:
alright, then no problem!

tin1254, 2020-06-19:
> I don't think this will break dataset replay. @tin1254 can you please check if the german and chinese merging maps work?

I found a bug running on those maps, I will try this after fixing the bug

tin1254, 2020-09-12:
> I don't think this will break dataset replay. @tin1254 can you please check if the german and chinese merging maps work?

chinese works but german doesn't, it's because of the coordinate values of the map are too large

closed 2020-10-20 by klemense1