bark-simulator/bark #379: macOS pip package build

patrickhart, 2020-06-18:
Does not work running `bash package.sh`.
What further steps would be necessary for the package build to work on macOS?

mansoorcheema, 2020-06-19:
@patrickhart I don't have mac so i am not sure about the exact cause but it seems its due to the dummy native C++ extension module that was added in setup.py for marking the package as platform dependent. I have discussed it in the packaging guide document. A brief idea is that the bark library (C++) is not created by setup.py so to mark package as platform dependent either the library has to be compiled using setup.py or has to be specified the package as native some other way. Its kind of patchy but not much can be done. There is --plat tag alternative that can be used for wheel building but that ignores cython version and more platform specific details. Anyways, I have committed a fix that you check and let me know.

patrickhart, 2020-06-19:
@mansoorcheema thank you! I uploaded a wheel to the PyPI website and installed it locally, seems work fine! What does this platform dependency imply? With the provided patch, it should work on linux and mac? (even if it is just ubuntu and mac 64 it would be sufficient for now)

mansoorcheema, 2020-06-19:
Compiled bark C++ bark library core.so is platform dependent. As per current implementation the package shall work on both macos and linux. For macos as the wheel name suggests (bark_simulator-0.1.0-cp37-cp37m-macosx_10_15_x86_64.whl ) the pip package shall be installed only if the user meets the build requirements (macos 10.5, cpython3.7 and python 3.7)that were used to generate the package in order to have compatibility. Pip handles which wheel to pick for a particular machine so we should have a wheel for every version we want to support (in case of macos). The situation is a bit complicated for linux so I built a fallback wheel without any platform restriction that was compiled on ubuntu 64. pip should prefer the most suitable wheel and if it does not find it would install the default one. It can also pick the default linux compiled wheel for mac os if it does not find a suitable macos wheel (which obviously wont work because the shared library is compiled for linux platform), so to avoid this, a standard practice is to build a wheel for each supported configuration type. Its described in the packaging guide and if you need more detail, i can email you more.

patrickhart, 2020-06-19:
Thank you very much for the detailed answer. I will have another look at the packaging guide but as you already discussed it with Julian, I think he is up-to-date on that matter.

Thus, for now I will close this issue.

closed 2020-06-19 by patrickhart