bark-simulator/bark #395: Fail to fetch 'benchmark_database_release'

jkwwwwow, 2020-07-07:
Hello, I don't know if this [URL  for BARK-DB](https://github.com/bark-simulator/bark-databasse/) is still valid, cause I got the following error when trying to use `bazel test //...` to validate that BARK is working:
```
WARNING: Download from https://github.com/bark-simulator/benchmark-database/releases/download/2.0/benchmark_database_2.0.zip failed: class javax.net.ssl.SSLException Read timed out
ERROR: An error occurred during the fetch of repository 'benchmark_database_release':
   java.io.IOException: Error downloading [https://github.com/bark-simulator/benchmark-database/releases/download/2.0/benchmark_database_2.0.zip] to /home/user/.cache/bazel/_bazel_user/1e3271b9b9137d3d6b2b33ab813c3a74/external/benchmark_database_release/benchmark_database_2.0.zip: Read timed out
INFO: Repository ghr instantiated at:
  no stack (--record_rule_instantiation_callstack not enabled)
Repository rule http_archive defined at:
  /home/user/.cache/bazel/_bazel_user/1e3271b9b9137d3d6b2b33ab813c3a74/external/bazel_tools/tools/build_defs/repo/http.bzl:336:31: in <toplevel>
ERROR: /home/user/bark/examples/BUILD:3:8: //examples:benchmark_database depends on @benchmark_database_release//:v2.0 in repository @benchmark_database_release which failed to fetch. no such package '@benchmark_database_release//': java.io.IOException: Error downloading [https://github.com/bark-simulator/benchmark-database/releases/download/2.0/benchmark_database_2.0.zip] to /home/user/.cache/bazel/_bazel_user/1e3271b9b9137d3d6b2b33ab813c3a74/external/benchmark_database_release/benchmark_database_2.0.zip: Read timed out
ERROR: Analysis of target '//examples:benchmark_database' failed; build aborted: Analysis failed
INFO: Elapsed time: 227.965s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (0 packages loaded, 0 targets conf\
FAILED: Build did NOT complete successfully (0 packages loaded, 0 targets conf\
igured)
```
Or is this just a problem with my network?

klemense1, 2020-07-08:
I just tried it out, could not reproduce it. 
Could you retry?
Also, can you checkout the repository itself manually? https://github.com/bark-simulator/benchmark-database
If that does not help, which Bazel version are you using?

klemense1, 2020-07-17:
@jkwwwwow does this problem still exist?

patrickhart, 2020-08-24:
Closed due to inactivity.

closed 2020-08-24 by patrickhart