bark-simulator/bark #446: GCC internal compiler error

Prasheel24, 2020-10-25:
Hello team,

Great work!

I would like to know if there are any specific system requirements for installing the bark simulator?

I am trying to install it on Ubuntu16.04 on an 8GB ram system with 60GB free space. The compiler gcc version is 7.5.0 and bazel is at 3.7.9.

I get this internal compiler error with gcc. Hope the screenshots help!
[image]
[image]

Please do let me know if there is a most stable release that can be used to start working with.

Thanks.
Prasheel

klemense1, 2020-10-26:
Hi Prasheel, 
I am using bazel version 3.6.0 and gcc 7.5.0 on Ubuntu 18.04. However, my guess is you are running out of memory. 

Can you try to run it with the following options `bazel build //... --local_ram_resources=HOST_RAM*.4 --local_cpu_resources=2`
Best
Klemens

Prasheel24, 2020-10-27:
Thanks Klemens.! It worked! 

I had previously upgraded to gcc8 (suggested by patrick) and tried to run, it didn't work. So, I downgraded to 7.5 and my system crashed. Post recovery, I worked with gcc 7.5 and using less resources command, it works.!

Will be in touch for future!

closed 2020-10-27 by Prasheel24

patrickhart, 2020-10-27:
Sorry for the inconvenience!
Added the issue to the FAQs: https://bark-simulator.github.io/tutorials/