bark-simulator/bark #447: Failed to run examples:interaction_dataset

valaxkong, 2020-10-27:
The following are the outputs after executing `bazel run //examples:interaction_dataset`
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "/root/.cache/bazel/_bazel_root/c5dcd1083c8a09da41f8d6ed6da1bc83/execroot/bark_project/bazel-out/k8-fastbuild/bin/examples/interaction_dataset.runfiles/bark_project/examples/interaction_dataset.py", line 42, in <module>
    world_state.DoExecution(sim_step_time)
AttributeError: 'bark.core.world.World' object has no attribute 'DoExecution'.

klemense1, 2020-10-27:
Sorry for the inconvenience, there was a bug in the implementation (stemming from renaming some time ago). I just fixed it, it should work now.

However, please note that we are just shipping dummy files of the dataset. If you want to use the real dataset, you need to apply for it at http://interaction-dataset.com/ and then replace our dummy track files with the original ones.

valaxkong, 2020-10-27:
@klemense1 Thanks a lot! Besides that, do you have plans to introduce any other dataset, e.g. nuScenes，Argoverse, in this project?

klemense1, 2020-10-27:
@valaxkong you're welcome! We would like to integrate them, however we require OpenDrive Map files and those datasets do not ship that. So for now it's not on top of our priority list...
But of cause contributions are very much welcome, so if you want to add them with BARK, we will definitely support you if you experience some issues.

closed 2020-10-27 by klemense1