bark-simulator/bark #475: Roadmap of this project

valaxkong, 2020-12-01:
Could you team show up your future roadmap about this project?

patrickhart, 2020-12-03:
We would want to improve the usability of BARK, integrate more elaborate behavior models and planners, and extend BARK towards a verification and validation (V&V) tool.

What is your interest using BARK for? Let us know and we can talk about it.

valaxkong, 2020-12-07:
@patrickhart I'm interested in using BARK as a V&V tool. Actually, I have developed a very similar V&V tool for internal use. I believe that a generally accepted V&V benchmark are of great significance to both industry and academia. BARK is promising to be a generally accepted benchmark. Thanks for your work!

patrickhart, 2020-12-07:
@valaxkong if you have any concrete idea what is missing or what should be added to the roadmap feel free to let us know. Of course, all kind of contributions to BARK are welcome.

valaxkong, 2020-12-07:
@patrickhart 
I have seen the carla-interface project. It use carla client to interaction with Carla. I think BARK should have a similar mechanism to  make testing easy. At present, I'm trying to build BARK client api to enable cosimulation with my system. Do you have some suggestions?

patrickhart, 2020-12-07:
The carla-interface repo can be seen as a starting point for building interfaces certainly. 

- I would suggest to implement and/or overload the BARK runtime by introducing an async function `ReceiveData(..)` that waits for data (such as e.g. object lists that are then added to the BARK world in each step).
- Once the BARK world has been filled with external data, one can use the behavior models in BARK and the world.Step(..) function to advance the world-state
- Additionally a function could broadcast the data, such as `StreamBARKData(..)` with which an external simulator could be updated -- depending on the operational mode.

valaxkong, 2020-12-08:
OK! I'll try to do it like you said.
One other thing, would you like to offer a web display feature to show what is happening in BARK?

patrickhart, 2020-12-08:
> OK! I'll try to do it like you said.
> One other thing, would you like to offer a web display feature to show what is happening in BARK?

In my opinion that certainly would be nice. We were thinking of using [streetscape/xiviz](https://avs.auto/) and if we have the resources to do so we might tackle that next year.

valaxkong, 2020-12-08:
I sincerely wish you have enough resource to keep maintaining and promoting this repo.

Would you like to add some comfort related evaluations?

patrickhart, 2020-12-08:
Sure, that would be a great addition for the evaluation and can be done rather quickly depending on the comfort metric.
If e.g. the comfort of driving is defined as having minimum jerk, this would also lead to safer driving maneuvers as the physical limits of the vehicle would be avoided.

And: We always welcome contributions to BARK. Generic interfaces would certainly be a great addition!

valaxkong, 2020-12-09:
I will create a pull request when I complete a initial version of general interfaces.

Minimun jerk is good enough to be the first metric to evaluation comfort of driving. Do you notice this research [SMARTS](https://github.com/huawei-noah/SMARTS)? It propounds a lot of valuable metrics.

patrickhart, 2020-12-11:
Yes, we are aware of the related projects in the field, but did not have find to have a closer look at it.
We mainly focused on analysing Carla, Sumo, and others.

valaxkong, 2020-12-15:
Im not very familiar with SUMO. And I wonder if SUMO can reproduce a scenario with some way. If the answer is yes, it will be greate to use SUMO to drive environment dynamic.

closed 2021-02-19 by patrickhart