bark-simulator/bark #490: How to install planner_mcts module?

KexianShen, 2021-02-27:
Hi, recently I have been exploring your excellent project. I can run examples with 'bazel run //bark/examples' according to [installation](https://github.com/bark-simulator/bark/blob/master/docs/source/installation.md) `Setup on Linux`. Problems I have are listed below:

1. Which decision making method is used when I run 'bazel run //bark/examples:highway'? How can I get the info?
2. How to install repo planner_mcts to main bark build? Or bark's python wrapper has already contained this sub-module, so I didn't find any relevant c++ files? I searched the issue list and found [bark-ml issue 84](https://github.com/bark-simulator/bark-ml/issues/84) and [bark-ml issue 77](https://github.com/bark-simulator/bark-ml/issues/77). However, there is no utils/install.sh in repo planner_mcts same as in the repo bark_ml. So I have no clue how to install it.
3. I tried 'bazel run //bark/examples:benchmark_database --define planner_uct=true', it returns as follows ([bark issue 376](https://github.com/bark-simulator/bark/issues/376)):
```
(venv) (bark_python_env) kexian@kexian-HP-ProBook-440-G6:~/repo/bark$ bazel run //bark/examples:benchmark_database --define planner_uct=true
INFO: Build option --define has changed, discarding analysis cache.
INFO: Repository planner_uct instantiated at:
  /home/kexian/repo/bark/WORKSPACE:19:15: in <toplevel>
Repository rule git_repository defined at
  /home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git.bzl:199:33: in <toplevel>
ERROR: An error occurred during the fetch of repository 'planner_uct':
   Traceback (most recent call last):
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git.bzl", line 181, column 30, in _git_repository_implementation
                update = _clone_or_update(ctx)
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git.bzl", line 36, column 20, in _clone_or_update
                git_ = git_repo(ctx, directory)
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git_worker.bzl", line 91, column 12, in git_repo
                _update(ctx, git_repo)
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git_worker.bzl", line 104, column 10, in _update
                reset(ctx, git_repo)
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git_worker.bzl", line 148, column 9, in reset
                _git(ctx, git_repo, "reset", "--hard", git_repo.reset_ref)
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git_worker.bzl", line 169, column 15, in _git
                _error(ctx.name, start + list(args), st.stderr)
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git_worker.bzl", line 190, column 9, in _error
                fail("error running '%s' while working with @%s:\n%s" % (command_text, name, stderr))
Error in fail: error running 'git reset --hard bde9ccc9750fad5764afd84987fcfe15ff97dfbe' while working with @planner_uct:
fatal: Could not parse object 'bde9ccc9750fad5764afd84987fcfe15ff97dfbe'.
ERROR: Error fetching repository: Traceback (most recent call last):
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git.bzl", line 181, column 30, in _git_repository_implementation
                update = _clone_or_update(ctx)
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git.bzl", line 36, column 20, in _clone_or_update
                git_ = git_repo(ctx, directory)
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git_worker.bzl", line 91, column 12, in git_repo
                _update(ctx, git_repo)
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git_worker.bzl", line 104, column 10, in _update
                reset(ctx, git_repo)
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git_worker.bzl", line 148, column 9, in reset
                _git(ctx, git_repo, "reset", "--hard", git_repo.reset_ref)
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git_worker.bzl", line 169, column 15, in _git
                _error(ctx.name, start + list(args), st.stderr)
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git_worker.bzl", line 190, column 9, in _error
                fail("error running '%s' while working with @%s:\n%s" % (command_text, name, stderr))
Error in fail: error running 'git reset --hard bde9ccc9750fad5764afd84987fcfe15ff97dfbe' while working with @planner_uct:
fatal: Could not parse object 'bde9ccc9750fad5764afd84987fcfe15ff97dfbe'.
ERROR: /home/kexian/repo/bark/bark/python_wrapper/BUILD:3:10: //bark/python_wrapper:core.so depends on @planner_uct//bark_mcts/python_wrapper:planner_uct in repository @planner_uct which failed to fetch. no such package '@planner_uct//bark_mcts/python_wrapper': error running 'git reset --hard bde9ccc9750fad5764afd84987fcfe15ff97dfbe' while working with @planner_uct:
fatal: Could not parse object 'bde9ccc9750fad5764afd84987fcfe15ff97dfbe'.
ERROR: Analysis of target '//bark/examples:benchmark_database' failed; build aborted: Analysis failed
INFO: Elapsed time: 9.755s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (0 packages loaded, 387 targets configured)
FAILED: Build did NOT complete successfully (0 packages loaded, 387 targets configured)
(venv) (bark_python_env) kexian@kexian-HP-ProBook-440-G6:~/repo/bark$ bazel run //bark/examples:benchmark_database --define planner_uct=true
INFO: Repository planner_uct instantiated at:
  /home/kexian/repo/bark/WORKSPACE:19:15: in <toplevel>
Repository rule git_repository defined at:
  /home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git.bzl:199:33: in <toplevel>
ERROR: An error occurred during the fetch of repository 'planner_uct':
   Traceback (most recent call last):
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git.bzl", line 181, column 30, in _git_repository_implementation
                update = _clone_or_update(ctx)
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git.bzl", line 36, column 20, in _clone_or_update
                git_ = git_repo(ctx, directory)
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git_worker.bzl", line 91, column 12, in git_repo
                _update(ctx, git_repo)
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git_worker.bzl", line 104, column 10, in _update
                reset(ctx, git_repo)
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git_worker.bzl", line 148, column 9, in reset
[... 7 lines cut ...]
ERROR: Error fetching repository: Traceback (most recent call last):
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git.bzl", line 181, column 30, in _git_repository_implementation
                update = _clone_or_update(ctx)
        File "/home/kexian/.cache/bazel/_bazel_kexian/e86d5390f4724b709d4d9568ac602320/external/bazel_tools/tools/build_defs/repo/git.bzl", line 36, column 20, in _clone_or_update
                git_ = git_repo(ctx, directory)
[... cut ...] r running 'git reset --hard bde9ccc9750fad5764afd84987fcfe15ff97dfbe' while working with @planner_uct:
fatal: Could not parse object 'bde9ccc9750fad5764afd84987fcfe15ff97dfbe'.
ERROR: Analysis of target '//bark/examples:benchmark_database' failed; build aborted: Analysis failed
INFO: Elapsed time: 7.608s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (0 packages loaded, 0 targets configured)
FAILED: Build did NOT complete successfully (0 packages loaded, 0 targets configured)
```

tinmodeHuang, 2021-02-27:
for the 1st question, you can find the  ```BehaviorLaneChangeRuleBased```used in ```bark/bark/examples/highway.py```.
for the 2nd one, you just need to enter the ``` git clone https://github.com/bark-simulator/planner-mcts.git``` in a terminal window. the planner-mcts , severing as one branch of the bark ecosystem,  depends on the master```bark```, and are linked with ```bark``` automatically after built.

KexianShen, 2021-02-28:
@tinmodeHuang, do i need to git clone planner-mcts into the bark file?

In addition, I find that the include path in `planner-mcts/src/mcts_state_single_agent.cpp` shows as below:
```c++
#include "modules/world/observed_world.hpp"
#include "modules/world/evaluation/evaluator_drivable_area.hpp"
#include "modules/world/evaluation/evaluator_collision_ego_agent.hpp"
#include "modules/world/evaluation/evaluator_goal_reached.hpp"
```
Do I have to rename file `bark` to `modules`?

juloberno, 2021-03-01:
Hi together,

nice to hear you are interested in trying out BARK. You observed correctly that the integration of MCTS repo is not yet compatible with the new layout of the BARK main repo. I plan to fix this in the next month.

If that is fixed, there is no need for separate cloning. It is all managed by bazel. 

We have another mcts variant build on top of the basic planner-mcts. It should be more up to date. @klemense1 Is the rules mcts ready to be used?

klemense1, 2021-03-01:
@juloberno is right, you can use https://github.com/bark-simulator/planner-rules-mcts instead of rules_mcts, it works with BARK's master.

KexianShen, 2021-03-01:
@juloberno @klemense1 thank you both for your reply. I'd like to confirm the install steps: 
1. git clone planner-rules-mcts into bark file
2. get into the env
3. run `bazel test //...`
Am I right? Thank you.

patrickhart, 2021-03-01:
Yes, it has a similar virtual environment / Bazel-workflow.
By including the planner-rules-mcts into BARK you can use it as a behavior model.
A behavior model follows always has to have a plan function similar as to described here: https://bark-simulator.github.io/tutorials/developing_behavior_models/

The behavior model for the planner-rules-mcts can be found here: https://github.com/bark-simulator/planner-rules-mcts/blob/master/src/behavior_rules_mcts.hpp

tinmodeHuang, 2021-03-02:
@patrickhart @klemense1 @juloberno Hi, could you have a look at [#9](https://github.com/bark-simulator/planner-mcts/issues/9)

KexianShen, 2021-03-02:
Finally, I build the planner-rules-mcts successfully.

1. `git clone git@github.com:bark-simulator/planner-rules-mcts.git`
2. `cd planner-rules-mcts`
2. `bash ./python/setup_test_venv.sh`
3. `bash ./python/into_test_venv.sh`
4. `bazel test //...`

In addition, if getting the error of `fatal error:ltdl.h: No such file or directory`, `sudo apt install libltdl-dev`

closed 2021-03-02 by KexianShen