bark-simulator/bark #491: seemingly lacking of some dependency - eigen

tinmodeHuang, 2021-03-02:
**Purpose**: install and use the module```planner-mcts```.
**Steps**:

1. git clone https://github.com/bark-simulator/planner-mcts.git
2. cd planner-mcts
3. bash ./python/setup_test_venv.sh
4. source ./python/into_test_venv.sh
5. bazel test //...

**Output in the last step**: 
```bash
Starting local Bazel server and connecting to it...
INFO: Repository com_github_eigen_eigen instantiated at:
  /home/tinmode/planner-mcts/WORKSPACE:82:13: in <toplevel>
Repository rule http_archive defined at:
  /home/tinmode/.cache/bazel/_bazel_tinmode/6f8dd349eff6a6a6951a9078c887ed57/external/bazel_tools/tools/build_defs/repo/http.bzl:336:31: in <toplevel>
WARNING: Download from http://bitbucket.org/eigen/eigen/get/3.3.4.tar.bz2 failed: class com.google.devtools.build.lib.bazel.repository.downloader.UnrecoverableHttpException GET returned 404 Not Found
ERROR: An error occurred during the fetch of repository 'com_github_eigen_eigen':
   Traceback (most recent call last):
	File "/home/tinmode/.cache/bazel/_bazel_tinmode/6f8dd349eff6a6a6951a9078c887ed57/external/bazel_tools/tools/build_defs/repo/http.bzl", line 111, column 45, in _http_archive_impl
		download_info = ctx.download_and_extract(
Error in download_and_extract: java.io.IOException: Error downloading [http://bitbucket.org/eigen/eigen/get/3.3.4.tar.bz2] to /home/tinmode/.cache/bazel/_bazel_tinmode/6f8dd349eff6a6a6951a9078c887ed57/external/com_github_eigen_eigen/temp14593907265138333511/3.3.4.tar.bz2: GET returned 404 Not Found
ERROR: Error fetching repository: Traceback (most recent call last):
	File "/home/tinmode/.cache/bazel/_bazel_tinmode/6f8dd349eff6a6a6951a9078c887ed57/external/bazel_tools/tools/build_defs/repo/http.bzl", line 111, column 45, in _http_archive_impl
		download_info = ctx.download_and_extract(
Error in download_and_extract: java.io.IOException: Error downloading [http://bitbucket.org/eigen/eigen/get/3.3.4.tar.bz2] to /home/tinmode/.cache/bazel/_bazel_tinmode/6f8dd349eff6a6a6951a9078c887ed57/external/com_github_eigen_eigen/temp14593907265138333511/3.3.4.tar.bz2: GET returned 404 Not Found
INFO: Repository pybind11 instantiated at:
  /home/tinmode/planner-mcts/WORKSPACE:32:13: in <toplevel>
Repository rule http_archive defined at:
  /home/tinmode/.cache/bazel/_bazel_tinmode/6f8dd349eff6a6a6951a9078c887ed57/external/bazel_tools/tools/build_defs/repo/http.bzl:336:31: in <toplevel>
INFO: Repository boost instantiated at:
  /home/tinmode/planner-mcts/WORKSPACE:79:11: in <toplevel>
  /home/tinmode/.cache/bazel/_bazel_tinmode/6f8dd349eff6a6a6951a9078c887ed57/external/com_github_nelhage_rules_boost/boost/boost.bzl:125:21: in boost_deps
Repository rule http_archive defined at:
  /home/tinmode/.cache/bazel/_bazel_tinmode/6f8dd349eff6a6a6951a9078c887ed57/external/bazel_tools/tools/build_defs/repo/http.bzl:336:31: in <toplevel>
ERROR: /home/tinmode/planner-mcts/python/BUILD:3:11: //python:planner_uct depends on @com_github_eigen_eigen//:eigen in repository @com_github_eigen_eigen which failed to fetch. no such package '@com_github_eigen_eigen//': java.io.IOException: Error downloading [http://bitbucket.org/eigen/eigen/get/3.3.4.tar.bz2] to /home/tinmode/.cache/bazel/_bazel_tinmode/6f8dd349eff6a6a6951a9078c887ed57/external/com_github_eigen_eigen/temp14593907265138333511/3.3.4.tar.bz2: GET returned 404 Not Found
ERROR: Analysis of target '//python:planner_uct' failed; build aborted: Analysis failed
INFO: Elapsed time: 4.609s
INFO: 0 processes.
FAILED: Build did NOT complete successfully (55 packages loaded, 392 targets configured)
FAILED: Build did NOT complete successfully (55 packages loaded, 392 targets configured)
    Fetching ...roject; Cloning 77d3efd13b508d6eb33f8fad1ba1665666529513 of https://github.com/jul\
oberno/mamcts
```

KexianShen, 2021-03-05:
@tinmodeHuang, if you are in China, please try with a VPN.

tinmodeHuang, 2021-03-07:
> @tinmodeHuang, if you are in China, please try with a VPN.

there is only one word on that website, namely "Repository eigen/eigen not found". 
I tried to modify the rules in WORKSPACE  to pull the package from the [URL](https://gitlab.com/libeigen/eigen/-/archive/3.3.4/eigen-3.3.4.tar.bz2) but failed as a result of knowing little about rule setting. Moreover, I wonder if you did it successfully.

KexianShen, 2021-03-08:
@tinmodeHuang, I tried several times and it worked. The build rule for `eigen` is defined in `planner-rules-mcts/util/deps.bzl` from line 86 to line 103.

tinmodeHuang, 2021-03-08:
I'm sorry for don't expressing myself clearly, I mean that the URL in info printed above isn't found for **```planner-mcts```**. even so, the module ```planner-mcts``` can also worked for you.

patrickhart, 2021-04-06:
I will close this issue as it seems not to be a BARK specific problem. If not resolved feel free to reopen the issue.

closed 2021-04-06 by patrickhart