bark-simulator/bark #494: How to take it into effect that some modifications in '''/bark/bark/runtime/viewer/viewer.py

tinmodeHuang, 2021-03-12:
I tried to display velocity of ego agent in the figure by adding the following lines while running '''tfa_gnn.py'''.
```python
        if debug_text:
            if eval_agent_ids is not None:
                self.drawText(position=(0.1, 1.0), text="velocity: {:.2f}".format(
                    world.agents[eval_agent_ids[0]].state[int(StateDefinition.VEL_POSITION)]), fontsize=14)
            self.drawText(position=(0.1, 0.9), text="Scenario: {}".format(
                scenario_idx), fontsize=14)
            self.drawText(position=(0.1, 0.95),
                          text="Time: {:.2f}".format(world.time), fontsize=14)
```
As a result, the modification make no difference. I managed to make it instead after modifying another '''viewer''' in '''.cache''', which comes out after building. that is not what I want.

patrickhart, 2021-03-15:
You need to set the parameters for the visualization correctly.  If you look at the very top of the viewer.py you can see the parameters being parse from the ParameterServer.

tinmodeHuang, 2021-03-15:
I'm sorry for don't explain myself clearly, the description have already been modified.

patrickhart, 2021-03-17:
> I tried to add the following lines in `/bark/bark/runtime/viewer/viewer.py` while running `tfa_gnn.py`.
> 
> ```python
>             if eval_agent_ids is not None:
>                 self.drawText(position=(0.1, 1.0), text="velocity: {:.2f}".format(
>                     world.agents[eval_agent_ids[0]].state[int(StateDefinition.VEL_POSITION)]), fontsize=14)
> ```
> 
> after modifying, the code block show as follow:
> 
> ```python
>         if debug_text:  # True by default
>             ###
>             if eval_agent_ids is not None:
>                 self.drawText(position=(0.1, 1.0), text="velocity: {:.2f}".format(
>                     world.agents[eval_agent_ids[0]].state[int(StateDefinition.VEL_POSITION)]), fontsize=14)
>             ###
>             self.drawText(position=(0.1, 0.9), text="Scenario: {}".format(
>                 scenario_idx), fontsize=14)
>             self.drawText(position=(0.1, 0.95),
>                           text="Time: {:.2f}".format(world.time), fontsize=14)
> ```
> 
> As a result, the modification make no difference. I managed to make it instead after modifying another `viewer.py` in `/.cache/bazel/_bazel_tinmode/0534ffbce5f0177dc35c2eb30e06c204/execroot/bark_ml/bazel-out/k8-fastbuild/bin/examples/tfa.runfiles/bark_project/bark/runtime/viewer/` , the file is generated after building, but that is not what I expect.

This should actually work, maybe try to have a look where in the code you added it and remove the if-condition?

patrickhart, 2021-03-22:
Sorry for the delayed answer. I think it would be best if you use VS-Code and setup the python debugging toolchain. Then you can set breakpoints and slowly debug the code, which functions are called, etc.

tinmodeHuang, 2021-03-22:
In fact, after modifying ```/bark/bark/runtime/viewer/viewer.py```, the same modification also appeared in ```/home/tinmode/.cache/bazel/_bazel_tinmode/eed35bca151b9786939218b1ff3c4a84/execroot/bark_project/bazel-out/k8-fastbuild/bin/bark/examples/merging.runfiles/bark_project/bark/runtime/viewer.py```, and therefore the velocity was displayed in the figure at the moment running ```bazel run //examples:merging```. However, the same modification don't have appeared in ```/.cache/bazel/_bazel_tinmode/0534ffbce5f0177dc35c2eb30e06c204/execroot/bark_ml/bazel-out/k8-fastbuild/bin/examples/tfa.runfiles/bark_project/bark/runtime/viewer/viewer.py```, that resulted in displaying no velocity  in the figure while running ```bazel run //examples:tfa_gnn```.

patrickhart, 2021-03-22:
That is probably due to the BARK version used that is defined in the dependencies. If you modify something in BARK and you want to use that in BARK-ML immediately it is better to use a local BARK version.

In the BARK-ML folder you can replace in the `utils/dependencies.bzl`
```
  _maybe(
    git_repository,
    name = "bark_project",
    branch = "master",
    remote = "https://github.com/bark-simulator/bark",
  )
```
with
```
  _maybe(
    native.local_repository,
    name = "bark_project",
    path = "/LOCAL/PATH/TO/BARK"
  )
```

tinmodeHuang, 2021-03-22:
I just have verified your conjecture, that is right. In addition, I want  to post another new issue to learn more about debugging

patrickhart, 2021-03-22:
As this issue seems resolved I will close it now.