bark-simulator/bark #498: cannot open the source file

tinmodeHuang, 2021-03-22:
In face of the errors, is there any advices to give me? thanks ahead of time.

[image]

patrickhart, 2021-03-23:
These errors are "just" VS-Code specific that cannot find the includes.

tinmodeHuang, 2021-03-23:
so VS code of specific version is to blame for these errors

patrickhart, 2021-03-23:
Not a specific version, but VS-Code is unable to locate the, e.g., boost source files. You can try to install Bazel extensions for VS-Code, but I am not sure if this will fix these issues.

tinmodeHuang, 2021-03-23:
there is still the errors after installing the Bazel 0.4 extension, did you developers also meet the same trouble that VS Code cannot locate these source files besides boost ?

[image]

patrickhart, 2021-03-23:
These problems are generated by cpplint as far as I know. Maybe you can try to find a fix somehow this way.

tinmodeHuang, 2021-03-24:
there are still the problems after installing successfully bark on another machine. the bark directory is opened by selecting file→open directory once VS Code is installed and configured with the extensions including bazel, cpplint, python and C++. I'm thinking about what's wrong in the process

patrickhart, 2021-03-24:
Under macOS (latest Catalina) and using VS-Code 1.54, I do not get these error messages shown. Do you have similar problems?  @klemense1 @juloberno

tinmodeHuang, 2021-03-24:
could you pay a little attention to #499? please

patrickhart, 2021-03-25:
I close this issue then as resolved as it seems a VS-Code/Linux specific error. If you find a fix for it, feel free to share it with us!

closed 2021-03-25 by patrickhart

tinmodeHuang, 2021-03-25:
in fact, the problem isn't resolved.