bark-simulator/bark #505: output is reassigned as the variable embeddings

tinmodeHuang, 2021-03-29:
I wonder what the 2nd statement in the snippet does, which have confused me for long.
```C++
    output, network_state = self._encoder(embeddings, training=training)  # its call func be called
    output = embeddings
```

patrickhart, 2021-03-29:
As this is BARK-ML specific please consider posting it there. Also, feel free to create a PR for changing it.
Best,
Patrick

closed 2021-03-29 by patrickhart

tinmodeHuang, 2021-03-29:
sorry for no additon to detail it, read the complement, please!

patrickhart, 2021-03-30:
BARK-ML can be found here: https://github.com/bark-simulator/bark-ml/tree/master/utils

related issues should be posted there

tinmodeHuang, 2021-03-30:
I don't understand it, you mean the question ought to be posted in the project bark-ml.