bark-simulator/bark #506: distance between the two agents is always zero

tinmodeHuang, 2021-03-30:
purpose: calculating the distances between any two agents.
the snippet modified in the ```graph_observer.py``` as shown below:
```c++
from bark.core.geometry import Distance, Point2d

    # add edges to all visible agents
    for index, agent in agents:
      nearby_agents = self._nearby_agents(
        center_agent=agent, 
        agents=agents, 
        radius=self._visibility_radius)

      if self._add_self_loops:
        adjacency_matrix[index, index] = 1

      for target_index, nearby_agent in nearby_agents:
        # if agent is world.ego_agent:
        if agent.shape is not None and nearby_agent.shape is not None:
          print('%-8.3f' % Distance(agent.shape, nearby_agent.shape), end='')
```
the way to do it shows as the last lines of statements above.
outputs in the terminal were all normal except that the distance is laways zero.

patrickhart, 2021-03-30:
The agent.shape is not transformed but always the base shape. Need to use `GetPolygonFromState(agent.state)`

tinmodeHuang, 2021-03-30:
besides, could you please have a look at #505

closed 2021-03-30 by tinmodeHuang