bark-simulator/bark #518: strange phenomenon

tinmodeHuang, 2021-04-10:
info printed as below:
```shell
I0410 13:30:18.583123 140075456821056 sac_runner.py:94] Training iterations 23000-24000 took 3026.243 seconds (avg. 3.026s / iteration).
I0410 13:30:26.928425 140075456821056 tfa_runner.py:123] The ego agent reached its goal. 
I0410 13:31:02.116097 140075456821056 tfa_runner.py:123] The ego agent reached its goal. 
I0410 13:31:08.224915 140075456821056 tfa_runner.py:123] The ego agent reached its goal. 
I0410 13:31:21.354181 140075456821056 tfa_runner.py:123] The ego agent reached its goal. 
I0410 13:31:41.352142 140075456821056 tfa_runner.py:123] The ego agent reached its goal. 
I0410 13:31:48.012688 140075456821056 tfa_runner.py:148] The agent achieved an average reward of -0.600, collision-rate of 0.72000, took on average 41.720 steps, and reached the goal  0.200 (evaluated over 25 episodes).
I0410 13:31:48.049430 140075456821056 common.py:1003] Saved checkpoint: /home/tinmode/bark-ml/logs/episode_duration:14s,replaybuffer:100_000,Tau:0.05,interval:3/ckpt-24001
I0410 13:31:48.049718 140075456821056 tfa_agent.py:84] Saved checkpoint for step 24001.
 0# bark::commons::SegfaultHandler(int) in /home/tinmode/.cache/bazel/_bazel_tinmode/0534ffbce5f0177dc35c2eb30e06c204/execroot/bark_ml/bazel-out/k8-fastbuild/bin/examples/tfa.runfiles/bark_project/bark/core.so
 1# 0x00007F65DB7D0040 in /lib/x86_64-linux-gnu/libc.so.6
 2# 0x00007F652FAC0DE4 in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/matplotlib/ft2font.cpython-37m-x86_64-linux-gnu.so
 3# 0x00007F652FAC1FBB in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/matplotlib/ft2font.cpython-37m-x86_64-linux-gnu.so
 4# 0x00007F652FAC2096 in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/matplotlib/ft2font.cpython-37m-x86_64-linux-gnu.so
 5# 0x00007F652FAC23FD in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/matplotlib/ft2font.cpython-37m-x86_64-linux-gnu.so
 6# 0x00007F652FAC2953 in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/matplotlib/ft2font.cpython-37m-x86_64-linux-gnu.so
 7# 0x00007F652FAC2FA4 in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/matplotlib/ft2font.cpython-37m-x86_64-linux-gnu.so
 8# 0x00007F652FAC3799 in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/matplotlib/ft2font.cpython-37m-x86_64-linux-gnu.so
 9# 0x00007F652FA51CD8 in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/matplotlib/ft2font.cpython-37m-x86_64-linux-gnu.so
10# 0x00007F652FAC3DEC in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/matplotlib/ft2font.cpython-37m-x86_64-linux-gnu.so
11# 0x00007F652FAC4325 in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/matplotlib/ft2font.cpython-37m-x86_64-linux-gnu.so
12# 0x00007F652FAC4832 in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/matplotlib/ft2font.cpython-37m-x86_64-linux-gnu.so
13# 0x00007F652FAC4EC5 in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/matplotlib/ft2font.cpython-37m-x86_64-linux-gnu.so
14# 0x00007F652FAC510E in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/matplotlib/ft2font.cpython-37m-x86_64-linux-gnu.so
15# 0x00007F652FA509E6 in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/matplotlib/ft2font.cpython-37m-x86_64-linux-gnu.so
16# 0x00007F652FA57391 in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/matplotlib/ft2font.cpython-37m-x86_64-linux-gnu.so
17# 0x00007F652FA3F37A in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/matplotlib/ft2font.cpython-37m-x86_64-linux-gnu.so
18# 0x00007F652FA43205 in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/matplotlib/ft2font.cpython-37m-x86_64-linux-gnu.so
19# _PyMethodDef_RawFastCallKeywords in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
20# 0x000000000054A9C0 in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
21# _PyEval_EvalFrameDefault in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
22# _PyEval_EvalCodeWithName in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
23# _PyFunction_FastCallKeywords in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
24# 0x000000000054A880 in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
25# _PyEval_EvalFrameDefault in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
26# _PyFunction_FastCallDict in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
27# _PyEval_EvalFrameDefault in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
28# _PyEval_EvalCodeWithName in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
29# _PyFunction_FastCallKeywords in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
30# _PyEval_EvalFrameDefault in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
31# _PyEval_EvalCodeWithName in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
32# _PyFunction_FastCallKeywords in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
33# 0x000000000054A880 in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
34# _PyEval_EvalFrameDefault in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
35# _PyEval_EvalCodeWithName in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
36# _PyFunction_FastCallDict in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
37# _PyEval_EvalFrameDefault in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
38# _PyEval_EvalCodeWithName in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
39# _PyFunction_FastCallDict in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
40# _PyEval_EvalFrameDefault in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
41# _PyEval_EvalCodeWithName in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
42# _PyFunction_FastCallKeywords in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
43# _PyEval_EvalFrameDefault in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
44# _PyEval_EvalCodeWithName in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
45# _PyFunction_FastCallKeywords in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
46# 0x000000000054A880 in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
47# _PyEval_EvalFrameDefault in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
48# _PyFunction_FastCallDict in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
49# _PyEval_EvalFrameDefault in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
50# _PyEval_EvalCodeWithName in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
51# _PyFunction_FastCallKeywords in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
52# _PyEval_EvalFrameDefault in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
53# _PyEval_EvalCodeWithName in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
54# _PyFunction_FastCallKeywords in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
55# 0x000000000054A880 in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
56# _PyEval_EvalFrameDefault in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
57# _PyEval_EvalCodeWithName in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
58# _PyFunction_FastCallKeywords in /home/tinmode/bark-ml/bark_ml/python_wrapper/venv/bin/python3
[... 209 lines cut ...]
  File "/home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/tf_agents/environments/batched_py_environment.py", line 166 in _step
  File "/home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/tf_agents/environments/py_environment.py", line 203 in step
  File "/home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/tf_agents/environments/tf_py_environment.py", line 314 in _step_py
  File "/home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/tf_agents/environments/tf_py_environment.py", line 214 in _execute
[... cut ...]
  File "/home/tinmode/.cache/bazel/_bazel_tinmode/0534ffbce5f0177dc35c2eb30e06c204/execroot/bark_ml/bazel-out/k8-fastbuild/bin/examples/tfa.runfiles/bark_ml/examples/tfa.py", line 89 in run_configuration
  File "/home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/absl/app.py", line 251 in _run_main
  File "/home/tinmode/bark-ml/bark_ml/python_wrapper/venv/lib/python3.7/site-packages/absl/app.py", line 303 in run
  File "/home/tinmode/.cache/bazel/_bazel_tinmode/0534ffbce5f0177dc35c2eb30e06c204/execroot/bark_ml/bazel-out/k8-fastbuild/bin/examples/tfa.runfiles/bark_ml/examples/tfa.py", line 100 in <module>
```

patrickhart, 2021-04-11:
Hello there,

what code base is that based on? The current master and in which cases does this error occur?
It seems like this is a segmentation fault, in which case there is most probably an access violation (e.g., on an object that has been deleted).

Best,

Patrick

tinmodeHuang, 2021-04-11:
let me think for a while

patrickhart, 2021-04-11:
What do you use the lane_id for?  At first sight, this function seems to be correct. 

I would recommend you to assert that the values are not null pointers, e.g., using `BARK_EXPECT_TRUE(lanecorridor != nullptr);`

Do you access anything with the retrieved lane_id? But in general debugging and setting assertions should help narrowing down the problem.

tinmodeHuang, 2021-04-12:
the answer to the first question is to determine Ids of lane with each agent in.

patrickhart, 2021-04-13:
The function looks good on first sight, but as I cannot see the full code and how you are using it, you have to try to debug it by yourself. You can also attach the python process to the c++.

Have a look here: https://bark-simulator.readthedocs.io/en/latest/debugging.html#debugging-c-and-python

tinmodeHuang, 2021-04-13:
I tried to debug it but was blocked by the problem below, and it is a custom function

[image]

patrickhart, 2021-04-14:
As I do not have access to any of your code, I cannot help you further at this point. I would suggest for you to add as many assertions as you can to figure out where exactly it fails.

tinmodeHuang, 2021-04-14:
all right

closed 2021-04-14 by patrickhart