bark-simulator/bark #521: spawn agents

tinmodeHuang, 2021-04-22:
a road with two lane, self._ds_min, self._ds_max = 7, 12
for left lane, self._s_min, self._s_max = 3, 44
for other one, self._s_min, self._s_max = 3, 26
there should are only 8 agents generated in observation of the world, but it is learned from the simplified output printed that 9 agents were generated.
```bash
{0: bark.agent.Agent, 1: bark.agent.Agent, 2: bark.agent.Agent, 3: bark.agent.Agent, 4: bark.agent.Agent, 6: bark.agent.Agent, 7: bark.agent.Agent}
{0: bark.agent.Agent, 1: bark.agent.Agent, 2: bark.agent.Agent, 3: bark.agent.Agent, 4: bark.agent.Agent, 5: bark.agent.Agent, 7: bark.agent.Agent, 8: bark.agent.Agent, 9: bark.agent.Agent}
Traceback (most recent call last):
  File "/home/tinmode/.cache/bazel/_bazel_tinmode/0534ffbce5f0177dc35c2eb30e06c204/execroot/bark_ml/bazel-out/k8-fastbuild/bin/examples/tfa_gnn.runfiles/bark_ml/examples/tfa_gnn.py", line 88, in <module>

...

  File "/home/tinmode/.cache/bazel/_bazel_tinmode/0534ffbce5f0177dc35c2eb30e06c204/execroot/bark_ml/bazel-out/k8-fastbuild/bin/examples/tfa_gnn.runfiles/bark_ml/bark_ml/environments/single_agent_runtime.py", line 56, in reset
    return self._observer.Observe(observed_world)
  File "/home/tinmode/.cache/bazel/_bazel_tinmode/0534ffbce5f0177dc35c2eb30e06c204/execroot/bark_ml/bazel-out/k8-fastbuild/bin/examples/tfa_gnn.runfiles/bark_ml/bark_ml/observers/graph_observer.py", line 122, in Observe
    edge_features[index, target_index, :] = self._extract_edge_features(agent, nearby_agent)
IndexError: index 8 is out of bounds for axis 1 with size 8
```
I have been confused about it after checked out the ```config_with_ease.py```to generate agents several times.
I'll appreciate it if you could make it clear.

patrickhart, 2021-04-22:
The agents are sampled in the distance, have a look at the `ConfigWithEase` scenario generation. 

In these two spots sampling is being applied: 

 - https://github.com/bark-simulator/bark/blob/1d5f788fa1a58d3d74c0b935b89a3764d5b5c837/bark/runtime/scenario/scenario_generation/config_with_ease.py#L133

- https://github.com/bark-simulator/bark/blob/1d5f788fa1a58d3d74c0b935b89a3764d5b5c837/bark/runtime/scenario/scenario_generation/config_with_ease.py#L92

If you want a deterministic scenario, I would recommend you to overwrite the `def position(self, world):` function that does not make use of sampling and a call to `self.ds()`. However, in terms of training varying scenarios are hugely beneficial to no overfit on a single problem.

tinmodeHuang, 2021-04-22:
I have already checked the scenario generation quite many times, the maximum number of agents generated shoule be 8 by calculating instead of 9

patrickhart, 2021-04-22:
how did you calculate this? I think only the rear point of the vehicle is used in the simple scenario generation (config with ease).

For more elaborate uses I would recommend using the ConfigurableScenarioGeneration.

tinmodeHuang, 2021-04-22:
the function ```position()``` of the``` LaneCorridorConfig```class define calculation of the number of agents to generate.
for left lane, self._s_min, self._s_max = 3, 44, with the minimum spacing ```self._ds_min=7```, the number of agents which can be generated in the range (3, 44) is 5.
for right lane, self._s_min, self._s_max = 3, 26, with the minimum spacing ```self._ds_min=7```, the number of agents which can be generated in the range (3, 26) is 5.

patrickhart, 2021-04-22:
I assume you are talking about https://github.com/bark-simulator/bark-ml/blob/b9b42ab8124ab31781573b6a804e762271ca4ce6/bark_ml/environments/blueprints/merging/merging.py#L65

Have a look at the values in the blueprint. The ranges have been slightly changed in the latest push to make the traffic more realitistic.

tinmodeHuang, 2021-04-22:
Yeap, the value of ```sef._ds_min```above is from ```ds_min``` in ```dense```mode

tinmodeHuang, 2021-04-23:
here starting positions is at ```self._s_min```, namely three, this is reason why the subtracter is 3 in the expression ```(26-3)//7=3``` instead of zero

patrickhart, 2021-04-23:
I cannot follow your calculation exactly. However, does it really matter if there are 8 or 9 vehicles generated?
Traffic also varies in reality by number and order. To trace the number during training and evaluation you could implement a vehicle count evaluator.

Will close this for now.

closed 2021-04-23 by patrickhart