bark-simulator/bark #541: Cannot find module "bark.core"

Rx-SUN, 2021-09-01:
When debugging the python code "intersection.py", I followed the instruction from the tutorial and I got the error "no module named bark". I guess it is caused by wrong path. Then I used 

```
import sys
sys.path.append("/home/myname/bark")
```

to add the path. But there is still error "no module named bark.core". This time I cannot find "bark.core". 

This is what I added in launch.json file based on the tutorial https://bark-simulator.readthedocs.io/en/latest/debugging.html
```
{
                "name": "Python: Current File",
                "type": "python",
                "request": "launch",
                "program": "${file}",
                "console": "integratedTerminal",
                "env": {
                  "PYTHONPATH": "${workspaceFolder}/bazel-bin/bark/examples/intersection.runfiles/__main__/python:${workspaceFolder}/bazel-bin/bark/examples/intersection.runfiles/__main__"
                }
              }
```

Could you please tell me how to fix this?

(I have installed BARK correctly based on the website 
https://github.com/bark-simulator/bark/blob/master/docs/source/installation.md
And the python file "intersection.py" runs well in terminal with commond "bazel run //bark/examples:merging")

patrickhart, 2021-09-06:
My (educated) guess would be: `bark.core` is always moved to the actual build folder by bazel. This is done via the genrule command.

Possible solution: Add `${workspaceFolder}/bazel-bin/bark/` to the path as well. @juloberno might be able to help you better here.

juloberno, 2021-09-07:
Hi,
either you take Patrick's solution and add paths manually or in the meanwhile, we have a solution to set PYTHONPATH automatically. Here is a PR that adds this debugging feature to the merging example. Just hit F5 to start debugging while having the merging.py active in vs code.

https://github.com/bark-simulator/bark/tree/debug_merging

We will also merge this into master when it is ready.

patrickhart, 2021-09-07:
Problem should be resolved as it is merged into the master branch.

closed 2021-09-07 by patrickhart