bark-simulator/bark #543: Cannot debug c++ and python

Rx-SUN, 2021-09-10:
Hello, I followed the instructions in https://bark-simulator.readthedocs.io/en/latest/debugging.html and I still failed jumping from python file to c++ file.

I successfully ran  `bazel build //... --compilation_mode=dbg` and `readelf --debug-dump=decodedline core.so`(I did not find bark.so so I copied core.so to bark folder). I then followed the 1.2.3.4 steps but I cannot jump to c++ file and debug.

Here is my launch.json file:

```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Python: Current File",
            "type": "python",
            "request": "launch",
            "program": "${file}",
            "console": "integratedTerminal",
            "justMyCode": false,
            "env": {
                "PYTHONPATH": "${workspaceFolder}/bazel-bark/tools/debug"
  }
        },
        {
            "name": "(gdb) Launch",
            "type": "cppdbg",
            "request": "launch",
            "program": "${workspaceFolder}/bazel-bin/bark/models/tests/behavior_motion_primitive_test",
            "args": [],
            "stopAtEntry": false,
            "cwd": "${workspaceFolder}",
            "environment": [],
            "externalConsole": false,
            "setupCommands": [
                {
                    "description": "Enable pretty-printing for gdb",
                    "text": "-enable-pretty-printing",
                    "ignoreFailures": true
                }
            ]
        },
        {
            "name": "(gdb) Attach",
            "type": "cppdbg",
            "request": "attach",
            "program": "/home/rx-sun/anaconda3/envs/bark_env/bin/python",
            "cwd" : "${workspaceFolder}",
            "processId": "${command:pickProcess}",
            "MIMode": "gdb",
            "sourceFileMap" : {"/proc/self/cwd/": "${workspaceFolder}"}
            }
    ]
}
```
[image]
[image]
[image]

patrickhart, 2021-10-01:
The `bark.so` file is located in the `bazel-bin`folder. The main thing to make debugging work is to add this path in the `launch.json` and to build the project in debug mode. The other thing is that you then can execute the file in question whilst being in that file and pressing `F5`.

klemense1, 2021-12-07:
Did this resolve the issue? Can we close it?

Rx-SUN, 2021-12-07:
> 

Yes, thank you very much !

klemense1, 2021-12-07:
Great! If you like our work, please star BARK to make it more visible to others.
I will close this issue then...

closed 2021-12-07 by klemense1