bark-simulator/bark #566: Collision in intersection . What caused it?

mrzheng8, 2021-12-16:
I have completed the installation of BARK and the configuration of the environment. But when I run the examples by "bazel run //bark/examples:intersection"  , I found that multiple agents often collide in the running results, and the whole running is not very smooth. I want to know why there is a collision in this example.
[image]
[image]

juloberno, 2022-01-05:
Hi,
Sorry for our late consideration. There were some delays due to the Christmas holidays. 

I looked at the example. The model BehaviorIntersectionRuleBased is not yet adapted to the newest geometry implementations.

I switched the example to use the model BehaviorIDMClassic. The IDM is a car-following model but can be adapted to intersections with careful parameter tuning. 
```
# at what maximum lateral distance between vehicle shapes are other vehicles considered in the IDM calculations: We use a high value for vehicles giving way and low value for vehicles taking way
param_server_give_way["BehaviorIDMClassic"]["MaxLatDifferenceToBeFront"] = 15.0

# at what minimum longitudinal distance are other vehicles considered in the IDM calculations: We can avoid vehicles blocking the intersection by increasing this value. However, a to high value increases the risk of collisions
param_server_give_way["BehaviorIDMClassic"]["MaxLonDifferenceToBeFront"] = 4.0

# By increasing the minimum spacing parameter of the IDM vehicles brake such that the intersection area is not blocked
param_server_give_way["BehaviorIDMClassic"]["MinimumSpacing"] = 8.0
```

You can adapt these parameters to your use case.

mrzheng8, 2022-01-09:
When the "step_time=0.2" , The whole operation looks very stuck. I tried to change this number to 1 and 0.05, but the effect was not very good. How can I solve it?