bark-simulator/bark #568: bazel can'

ppjtzjb, 2022-01-11:


ppjtzjb, 2022-01-11:
Hello, when I try to verify the installation with bazel test //.... Fetching boost always fails.
Outputs:
in _http_archive_impl
                download_info = ctx.download_and_extract(
Error in download_and_extract: java.io.IOException: Error downloading [https://boostorg.jfrog.io/artifactory/main/release/1.76.0/source/boost_1_76_0.tar.gz] to /home/ahj2szh/.cache/bazel/_bazel_ahj2szh/4cf193a969e04826f42803cb5341d947/external/boost/temp2005063288673510406/boost_1_76_0.tar.gz: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target

Any ideas?

juloberno, 2022-01-20:
Hi,

we had never faced this error before. It seems like a certification issue of java as e.g., described when using bazel here: https://stackoverflow.com/questions/68731040/bazel-remote-cache-not-working-with-certificates-tls

closed 2022-02-23 by juloberno