bark-simulator/bark #572: can't pickle bark.core.models.behavior.BehaviorModel objects

flxebert, 2022-02-14:
I'm trying to reproduce the [developing behavior model tutorial](https://bark-simulator.github.io/tutorials/developing_behavior_models) in Ubuntu 18.04 and conda install.
In both the *pip install* and the *source* installation execution fails with the error 
`can't pickle bark.core.models.behavior.BehaviorModel objects`.

I couldn't find any documentation or issue related to that topic. Do I miss something? How can I fix this?

Here is the full output
[output.txt](https://github.com/bark-simulator/bark/files/8059666/output.txt)

juloberno, 2022-02-23:
The tutorial code is similar to the test cases implemented in bark/models/tests/py_behavior_model_test.py

You can have a look there and see if your implementation follows the test implementation. If errors persist, please add your code here and describe your build and execution pipeline.

flxebert, 2022-02-25:
The problem occurs if the derived behavior model is returned from the _controlled_behavior_model()_ in the _CustomLaneCorridorConfig_ class. The implementation of _DerivedBehaviorModel_ follows the ones in bark/models/tests/py_behavior_model_test.py.
I followed the installation tutorial, working in a conda environment with python 3.7 and installed bark via `pip install bark-simulator`.

Ubuntu 18.04
python 3.7.11
bark 1.4.9

Running the following code (which comes from the tutorial) leads to the _pickle_ error

```python
import numpy as np
from bark.core.world.opendrive import *
from bark.core.world.goal_definition import *

from bark.core.models.behavior import BehaviorModel
from bark.core.models.behavior import BehaviorMobilRuleBased

from bark.runtime.runtime import Runtime
from bark.examples.paths import Data
from bark.runtime.commons.parameters import ParameterServer
from bark.runtime.viewer.matplotlib_viewer import MPViewer
from bark.runtime.scenario.scenario_generation.config_with_ease import \
  LaneCorridorConfig, ConfigWithEase

params = ParameterServer()

class DerivedBehaviorModel(BehaviorModel):
  def __init__(self, params=None):
    super(DerivedBehaviorModel, self).__init__(params)

  def Plan(self, step_time, observed_world):
    super(DerivedBehaviorModel, self).ActionToBehavior(np.array([2., 1.], dtype=np.float32))
    trajectory = np.array([[0., 0., 0., 0., 0.], [0., 0., 0., 0., 0.]], dtype=np.float32)
    super(DerivedBehaviorModel, self).SetLastTrajectory(trajectory)
    return trajectory

  def Clone(self):
    return self

class CustomLaneCorridorConfig(LaneCorridorConfig):
  def __init__(self, params=None, **kwargs):
    super(CustomLaneCorridorConfig, self).__init__(params, **kwargs)

  def goal(self, world):
    road_corr = world.map.GetRoadCorridor(self._road_ids, XodrDrivingDirection.forward)
    lane_corr = self._road_corridor.lane_corridors[0]
    return GoalDefinitionPolygon(lane_corr.polygon)

  def controlled_behavior_model(self, world):
    return DerivedBehaviorModel(self._params)

# configure both lanes
left_lane = CustomLaneCorridorConfig(params=params,
                                     lane_corridor_id=0,
                                     road_ids=[0, 1],
                                     behavior_model=BehaviorMobilRuleBased(params),
                                     s_min=0.,
                                     s_max=50.)

# this lane has controlled_ids; ego vehicle is placed on this lane
right_lane = CustomLaneCorridorConfig(params=params,
                                      lane_corridor_id=1,
                                      road_ids=[0, 1],
                                      controlled_ids=True,
                                      behavior_model=BehaviorMobilRuleBased(params),
                                      s_min=0.,
                                      s_max=20.)

# finally, generate 3 scenarios on the merging map
scenarios = \
  ConfigWithEase(
    num_scenarios=3,
    map_file_name=Data.xodr_data("DR_DEU_Merging_MT_v01_shifted"),
    random_seed=0,
    params=params,
    lane_corridor_configs=[left_lane, right_lane])

viewer = MPViewer(params=params,
                  x_range=[-35, 35],
                  y_range=[-35, 35],
                  follow_agent_id=True)
env = Runtime(step_time=0.2,
              viewer=viewer,
              scenario_generator=scenarios,
              render=True)

# run 3 scenarios
for _ in range(0, 3):
  env.reset()
  # step scenario 90 time-steps
  for step in range(0, 90):
    env.step()
```