bark-simulator/bark #573: Cannot find the "Params" class

mrzheng8, 2022-02-21:
[image]
The figure above shows the "parameters" module. You can see from the third line that the "Params" class is imported as the parent class, but I can't find the location of this "Params" class.
Where can I find the "Params" class? I would appreciate it if you could answer my question.

juloberno, 2022-02-23:
What code are you running? Did you follow the installation instructions in the Readme?

CesarLiu, 2022-03-03:
> [image] The figure above shows the "parameters" module. You can see from the third line that the "Params" class is imported as the parent class, but I can't find the location of this "Params" class. Where can I find the "Params" class? I would appreciate it if you could answer my question.

Hi,I think this class "Params" is written in cpp, you could refer to https://github.com/bark-simulator/bark/blob/master/bark/commons/params/params.hpp