bark-simulator/bark #578: How can the new model be found？

mrzheng8, 2022-04-11:
I copied the MCTs model to the behavior folder, modified the dependencies in the build file and the model in the example, but couldn't find the model after running. Is this a compilation problem? How can the C + + file of the new model be found?
Errors are reported as follows：
project/bark/examples/mctsceshi.py", line 35, in <module>
    behavior_model=BehaviorUCTSingleAgent(param_server_give_way),
NameError: name 'BehaviorUCTSingleAgent' is not defined

mrzheng8, 2022-04-11:
Error using bazel build. I think there may be a problem with the dependence of this model?

mrzheng8, 2022-04-11:
I try to use the https://github.com/bark-simulator/planner-rules-mcts models.
But I found only /src/behavior_rules_mcts.hpp, there is no corresponding cpp file.

mrzheng8, 2022-04-12:
I have downloaded  https://github.com/bark-simulator/planner-rules-mcts  .
But its build file does not compile the /src/behavior_rules_mcts.hpp.
How does it compile?

mrzheng8, 2022-04-13:
I copied the MCTs model to the behavior folder, modified the dependencies in the build file and the model in the example, but couldn't find the model after running. Is this a compilation problem? How can the C + + file of the new model be found?
Errors are reported as follows：
project/bark/examples/mctsceshi.py", line 35, in
behavior_model=BehaviorRulesMcts(param_server_give_way),
NameError: name 'BehaviorRulesMcts' is not defined

mrzheng8, 2022-04-13:
1.I use bazel build to compile the behavior_rules_mcts. //There are only HPP files. Create a corresponding cpp file.
2.In python file, from bark.core.models.behavior import *. //I think the compiled executable model is being imported at this time.
3.Run the python file, and an error occurred.//NameError: name 'BehaviorRulesMcts' is not defined

mrzheng8, 2022-04-21:
The following is the error message：

behavior_model=BehaviorRulesMctsUct(param_server_give_way),
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. bark.core.models.mcts.BehaviorRulesMctsUct(arg0: bark.core.commons.Params, arg1: bark::world::prediction::PredictionSettings, arg2: List[bark::world::evaluation::BaseLabelFunction], arg3: List[ltl::RuleMonitor], arg4: Dict[int, List[ltl::RuleMonitor]])

mrzheng8, 2022-04-25:
The following error occurred when I replaced the model name. I took a part of it.
Why did this mistake happen? Is there an error in the header file mvmcts.h:79

WARNING: Logging before InitGoogleLogging() is written to STDERR
I20220425 21:56:11.574766  1561 mvmcts.h:79] Max search samples: 2000
I20220425 21:56:11.574784  1561 mvmcts.h:85] starting state: 
 0# bark::commons::SegfaultHandler(int) in /home/su/.cache/bazel/_bazel_su/61c2ccfa09a92e90cc5125978445bf56/execroot/planner_rules_mcts/bazel-out/k8-fastbuild/bin/python/mctsceshi.runfiles/bark_project/bark/core.so
 1# 0x00007FD1F2829F10 in /lib/x86_64-linux-gnu/libc.so.6
 2# bark::models::behavior::MvmctsState::GetNumActions(unsigned char) const in /home/su/.cache/bazel/_bazel_su/61c2ccfa09a92e90cc5125978445bf56/execroot/planner_rules_mcts/bazel-out/k8-fastbuild/bin/python/mctsceshi.runfiles/bark_project/ba
...
...
37# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
38# 0x00005587605C7773 in /home/su/planner-rules-mcts-master/python/venv/bin/python3

terminate called without an active exception
已放弃 (核心已转储)

CesarLiu, 2022-06-08:
Hi,
I think it' not necessary to put mcts in bark, and you can load it by configuration in bazel dependencies, like what has been done in here: https://github.com/bark-simulator/example_benchmark#behavior-models

best,